Logicademy/HDLGen-ChatGPT #58: Project Folder Path Reset Due To Environment Change

Teehee2, 2023-11-06:
## Issue Type: Bug/Mouldy Feature
### Description:
Changing project environment once in an opened project will reset the project folder path to the same path as the newly selected project environment. This can cause issues especially for users with long paths, where they may not notice the change. Can cause new project folders to be created where the environment folder is specified.
### Proposed fix:
Add checks to verify if existing project folder is within a directory of the environment folder, if it is, do not change the project folder, if not, allow the project folder path to reset to the environment path.

### HDLGen-ChatGPT version: 
Python 11/10/2023
### System specification:
Windows 11 Pro Version	10.0.22635 Build 22635 x64
Intel Core i7-1065G7
16GB RAM
Python 3.10.7

Daxorinator, 2023-11-07:
ACK - We're aware of this issue and have written fixes for it which will come in the next release. They are currently being tested.

Daxorinator, 2024-05-22:
Resolved by #77

closed 2024-05-22 by Daxorinator